ros-controls/ros2_controllers #1841: Estimate position using rolling mean velocity

Narukara, 2025-07-30:
While testing on an Ackerman vehicle, I found that the `velocity_rolling_window_size` setting only affects the velocity estimated by the odometry, not the position. It seems to be caused by the following code:

https://github.com/ros-controls/ros2_controllers/blob/b21a7e1d296f5176538f874693de7472cda7fbf7/steering_controllers_library/src/steering_odometry.cpp#L56-L76

Is this by design? I think this design seems counterintuitive. I would expect the odometry to use filtered velocity to estimate position.

christophfroehlich, 2025-08-27:
I think the documentation of the parameter is clear:
`The number of velocity samples to average together to compute the odometry twist.linear.x and twist.angular.z velocities.`

This follows the same idea as for the mother of controllers here, the diff_drive_controller.

I'd say that integration has already a filtering effect (integrating zero mean additive white gaussian noise would result in zero output), and therefore the filtered variable is not needed here. 

What you are observing here is the difference of wheel speed and vehicle speed due to slipping, right? You "fixed" that by filtering the wheel velocity, but this is not what this parameter is supposed for: You compensated a physical property with some filtering, but this will only work in simulation where the slippage is always the same.

christophfroehlich, 2025-08-27:
If you need accurate odometry for a real robot, you will need other means of localization on top of wheel odometry (LIDAR SLAM, GNSS, IMU EKF, whatsoever)..

Narukara, 2025-08-27:
Thanks! I understand the reasoning behind this design now.

closed 2025-08-27 by Narukara