ros-controls/ros2_controllers #1937: Unified the unit of `cmd_vel_timeout`

Narukara, 2025-09-30:
**Describe the bug**

Several controllers have a `cmd_vel_timeout` parameter. However, their units are inconsistent. Specifically, I found these:

| parameter                                                    | unit         | type   |
| ------------------------------------------------------------ | ------------ | ------ |
| [tricycle_controller_parameter.yaml](https://github.com/ros-controls/ros2_controllers/blob/dca8f45ea8b96088bbeb9e54dc963eea087ebe50/tricycle_controller/src/tricycle_controller_parameter.yaml#L69) | milliseconds | int    |
| [omni_wheel_drive_controller_parameters.yaml](https://github.com/ros-controls/ros2_controllers/blob/dca8f45ea8b96088bbeb9e54dc963eea087ebe50/omni_wheel_drive_controller/src/omni_wheel_drive_controller_parameters.yaml#L93) | seconds      | double |
| [diff_drive_controller_parameter.yaml](https://github.com/ros-controls/ros2_controllers/blob/dca8f45ea8b96088bbeb9e54dc963eea087ebe50/diff_drive_controller/src/diff_drive_controller_parameter.yaml#L94) | seconds      | double |

This is a bit confusing. Maybe we should unify their units.

Additionally, two controllers are using `reference_timeout`. This appears to be the same as `cmd_vel_timeout`. Maybe the parameter names should be unified as well?

`reference_timeout`: [steering_controllers_library.yaml](https://github.com/ros-controls/ros2_controllers/blob/dca8f45ea8b96088bbeb9e54dc963eea087ebe50/steering_controllers_library/src/steering_controllers_library.yaml#L2), [mecanum_drive_controller.yaml](https://github.com/ros-controls/ros2_controllers/blob/dca8f45ea8b96088bbeb9e54dc963eea087ebe50/mecanum_drive_controller/src/mecanum_drive_controller.yaml#L2)

**Expected behavior**
Consistent units and types.

**Environment:**
 - Version: master branch

ielson, 2026-03-26:
So I have checked here and from my research, these parameters are used for the same kind of behavior:. The main difference todayis API naming:

  - cmd_vel_timeout is used by controllers subscribing to ~/cmd_vel
  - reference_timeout is used by controllers subscribing to ~/reference
 The types in them are the same(`geometry_msgs/msg/TwisStamped`)

None of them are used when on chained mode, so following @christophfroehlich answer I would call them all reference_timeout

> You are right about the naming. reference_timeout would be more generic, but I'm not sure at the moment if this logic is also used if the controllers are in chained mode.

  Proposed plan:

- Standardize timeout units and type across all affected controllers to seconds as double
- Unify the parameter name across controllers to be reference_timeout
- If compatibility is important, keep the old parameter names as deprecated aliases for a transition
  period
- Update parameter definition files, code paths, tests, example configs, and user documentation
  together

christophfroehlich, 2026-05-15:
when changing ms to seconds, we could also print a warning when the value is bigger than 100 or so, maybe automatically use the value/1000 then.

christophfroehlich, 2026-10-18:
@ielson please consider my last comments here if you still are able to work on this issue